riuson/lcd-image-converter #37: Сглаживание включается для некоторых размеров шрифта без спроса

DmitryBondarenko, 2015-01-27:
Добрый день!

Ревизия ff449e0 от 2014-12-14 12:47:27 +0500

Создаю новый шрифт, задаю настройки преобразования "Монохромный, порог"

Задаю параметры шрифта - моноширинный, сглаживание выключено, размер шрифта 16 и менее - все работает как ожидается.

Меняю параметры, изменив размер шрифта больше 16, но меньше 72 - в битмапах шрифта появляется сглафивание полутонами, которое я не включал

Меняю параметры, изменив размер шрифта больше 72 - снова все ОК.

В версии Ревизия a7e11ff от 2014-07-19 18:41:19 +0600 такого глюка нет.

Спасибо за отличную программу!
С уважением, Дмитрий

riuson, 2015-01-27:
Здравствуйте!

Изменение, ставшее причиной такого поведения:
https://github.com/riuson/lcd-image-converter/commit/85487685fa34b7555463ecfbd9c27c663b6aae45
По запросу пользователя:
https://github.com/riuson/lcd-image-converter/issues/30

Это не антиалиасинг, поэтому эта опция не влияет на результат.
До того изменения символы рисовались на RGB32, без альфа-канала. Округление значения пиксела до ближайшего выполнялось каким-то встроенным методом.
Можете изменить это значение в коде и получить другие результаты, может даже лучше.
Если выбрать Monochrome Edge 181, будет что-то аналогичное июльскому, см. скриншот:

[image]

closed 2015-02-08 by riuson

riuson, 2015-11-03:
Альфа-канал сделан опциональным в 9a6644ea6973d4d1fd5843b98f136044b6932ebe 2015-11-03

DmitryBondarenko, 2015-11-03:
Владимир, спасибо за нововвседение! Очень полезная фунция, если работаешь с маленькими монохромными дисплеями.