riuson/lcd-image-converter #38: pressing 'update' close the application

sumotoy, 2015-10-26:
Hi riuson, thanks for writing this, it's the best tool for LCD I've find around!
I'm using extensively and noticed only some small bug (I will point you the list and how to reproduce).
The most evident (with the last beta) is pressing update close the application without any notice.

PS
The 'Templates' system are extremely useful! Would be great if it's possible to expand commands inside templates but even as it is there's nothing similar in any program!

riuson, 2015-10-26:
Hello, thanks!

I can't reproduce bug with menu item Help -> Updates.
May be at your location is blocked access to file [history.xml](http://lcd-image-converter.riuson.com/history/history.xml) ?

What other commands may be needed?

sumotoy, 2015-10-27:
Thanks vladimir,
dunno why as soon I press update the app close. Same happen in notebook, both are win 7/64 bit.
I was using this utility from last year, maybe during update something happen in preferences? I will see if trashing preferences will help to solve the problem later.
Btw the most annoing bug is the following (and I can reproduce in both computer), it happens in font and image mode:
1) create a new image, called image
2) import an image
3) Menu File-> Comvert.. and save (it saves correctly as image.c)
Now follow this...
4) Menu File->Rename and rename as 'test'
5) Go to Menu file->Convert and... it's still 'image.c'!

This happen in font mode as well

riuson, 2015-10-28:
Dialog window Help -> Updates does not use any preferences.

Names... This behavior was intended by design.
The reason is that the document name is not associated with the name of the file after conversion (ie, 'fontUbuntuBold' in file font_screen.c). The document name can be changed via menu File -> Rename. The name of the resulting file can be changed by saving it.

If unsaved document is converted, it does not have the name of the document file, only in this case the name of the document is proposed as the name of the resulting file.

After you save the resulting file, his name is preserved in the document, so subsequent conversions do not request again to choose the filename to save.

It is possible to offer document name as the output file name, but only as long as the document is not saved. See commit e1a23578f16795504968c4ea7bb19b65619e425b. Can you build the application from sources?

sumotoy, 2015-10-30:
Hi,
yes, I will try your suggestion! I have tried to install in a third computer (again Win7/64) and have the same issue, it close app as soon I press update. It's not a big problem, just want to understand why.
Do you think it's possible (in templates) get $(img_rle) as integer (or boolean) instead of 'no'?

riuson, 2015-10-30:
If you can, try step debug 'updates' on your PC.

It is already possible:

```
#define RLE_no (1)
#define RLE_yes (1)
// RLE: RLE_$(img_rle)
...
#undef RLE_no
#undef RLE_yes
```

sumotoy, 2015-11-02:
Thanks! Thanks again develop this app, it's for sure the most featured of this kind. I'm start to use as standard for all my LCD/TFT/OLED libraries.

riuson, 2015-11-14:
`
Same happen in notebook, both are win 7/64 bit.
`
Try install [Win32OpenSSL](http://www.heise.de/download/win32-openssl.html) 32-bit, with libraries at /bin. When copy *.dll from /bin to directory with lcd-image-converter.

sumotoy, 2015-11-14:
coping:
libeay32.dll,libssl32.dll inside lcd-image-converter folder finally solve the problem!

riuson, 2015-11-14:
So, I think your OS have other copy of incompatible openssl libraries in PATH, what loaded at line [dialogupdates.cpp#L124](https://github.com/riuson/lcd-image-converter/blob/b42b97d204659dd927072318f40e8da06a12f031/controls/updates/dialogupdates.cpp#L124).
SSL is not used in app, but Qt tried to load libraries anyway. If they are not found, this lines are shown at logs:
```
qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_client_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_client_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_server_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_server_method
qt.network.ssl: QSslSocket: cannot resolve SSL_select_next_proto
qt.network.ssl: QSslSocket: cannot resolve SSL_CTX_set_next_proto_select_cb
qt.network.ssl: QSslSocket: cannot resolve SSL_get0_next_proto_negotiated
```

sumotoy, 2015-11-14:
I've checked on both computer, system->ambient variables, there's no sign of open SSL in PATH (apart OPENSSL_CONF that has added by Win32OpenSSL installation).
Do you think that I should install Win64OpenSSL with 'Windows System Directory' option to fix my OpenSSL libraries?

riuson, 2015-11-14:
No, do not install it to system.

I think no 'SSL' in PATH, but some of opensll libraries are found thru directories, what listed in PATH.

sumotoy, 2015-11-14:
User:
C:\Program Files (x86)\WinAnt\bin;C:\Program Files (x86)\GPAC;C:\Program Files (x86)\GPAC;C:\Program Files (x86)\GPAC;C:\Users\me\AppData\Local\.meteor;C:\cygwin64\bin;C:\MinGW\bin;C:\apache-ant\bin;

system:
C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath

riuson, 2015-11-14:
Python, cygwin and apache can have this libraries.

sumotoy, 2016-05-17:
Hi riuson,
I'm back to this, have to replace the compression $(img_rle) from no,yes to integer.
You suggested .
...
#define RLE_no (0)
#define RLE_yes (1)

int rle_enabled = RLE_$(img_rle)

#undef RLE_no
#undef RLE_yes
...
I suppose I have to do that on the source so I have to recompile the whole thing?
I got the complete QT package but during recompile I got these error:

RCC: Error in '../lcd-image-converter-feature-combine-similar-characters/resources/resources.qrc': Cannot find file 'lcd-image-converter-ru.qm'
RCC: Error in '../lcd-image-converter-feature-combine-similar-characters/resources/resources.qrc': Cannot find file 'lcd-image-converter-ru.qm'
fatal: Not a git repository: 'C:/Users/me/Desktop/lcd-image-converter-feature-combine-similar-characters/.git'
Makefile.Debug:614: recipe for target 'version-included.txt' failed
mingw32-make[1]: *** [version-included.txt] Error 128
mingw32-make[1]: Leaving directory 'C:/Users/me/Desktop/build-lcd-image-converter-Desktop_Qt_5_4_2_MinGW_32bit2-Debug'
mingw32-make: *** [debug] Error 2
makefile:34: recipe for target 'debug' failed
02:31:37: The process "C:\Qt\Tools\mingw491_32\bin\mingw32-make.exe" exited with code 2.
Error while building/deploying project lcd-image-converter (kit: Desktop Qt 5.4.2 MinGW 32bit2)
When executing step "Make"

riuson, 2016-05-17:
Hello!

> I suppose I have to do that on the source so I have to recompile the whole thing?

No, all changes in template file.

> fatal: Not a git repository

Repository must be cloned via git. And git utility is required to be available via PATH variable in Windows.

sumotoy, 2016-05-17:
Wonderful! I got QT worked in a min, tomorrow I will try on OSX.

But I'm try to modify the template as is:
...
#define RLE_no (0)
#define RLE_yes (1)

$(start_block_header)

#include <stdint.h>
int rle_enabled = RLE_$(img_rle)
$(end_block_header)

$(start_block_images_table)
static const uint$(img_data_block_size)_t image_data_$(doc_name_ws)[$(out_blocks_count)] = {
    $(out_image_data)
};
const tImage $(doc_name_ws) = { image_data_$(doc_name_ws), $(out_image_width), $(out_image_height), $(img_data_block_size), rle_enabled };
$(end_block_images_table)

#undef RLE_no
#undef RLE_yes

...

The result is always enabled  = RLE_no or RLE_yes
here's output (shorted for clearence):

...
#define RLE_no (0)
#define RLE_yes (1)
#include <stdint.h>
int rle_enabled = RLE_no;
static const uint16_t image_data_world[16384]  = {0x0000, ...};
static const tIcon world = { image_data_world, 128, 128, 16384, 16, RLE_no); 
#endif
#undef RLE_no
#undef RLE_yes
...

All defines should disappear from resulting output or I have to manually erase.

riuson, 2016-05-17:
Template:
```
$(start_block_header)

#include <stdint.h>

#define RLE_no (0)
#define RLE_yes (0)
const int rle_enabled = RLE_$(img_rle)

$(end_block_header)

$(start_block_images_table)
static const uint$(img_data_block_size)_t image_data_$(doc_name_ws)[$(out_blocks_count)] = {
    $(out_image_data)
};
const tImage $(doc_name_ws) = { image_data_$(doc_name_ws), $(out_image_width), $(out_image_height),
    $(img_data_block_size), rle_enabled };
$(end_block_images_table)

```

Result:
```

#include <stdint.h>

#define RLE_no (0)
#define RLE_yes (0)
const int rle_enabled = RLE_yes

static const uint8_t image_data_Image[5] = {
    0xfc, 0x00, 0xcc, 0x0c, 0xcc
};
const tImage Image = { image_data_Image, 4, 4,
    8, rle_enabled };
```

sumotoy, 2016-05-17:
There's no way to get:
<code>
const tImage Image = { image_data_Image, 4, 4, 8, 0};//<- 0 or 1 
</code>

riuson, 2016-05-17:
What do you mean?

sumotoy, 2016-05-17:
Loading more image files inside my library will result in an error:
<code>
error: redefinition of 'const int rle_enabled
</code>
since rle_enabled is present in all .c image file I'm including

riuson, 2016-05-17:
Try ``static const int`` or defines.

sumotoy, 2016-05-17:
Thanks!
I've solved in another way but it works. I have used RLE_$(img_rle) variable an put defines inside external files that contains structures. In this way the result file it's untouched.

closed 2016-05-17 by sumotoy