riuson/lcd-image-converter #39: null pointers for unused characters

0xb0ba, 2016-03-17:
not to add data (image_data_Font & tImage) to the unused characters (width and / or height = 0) to save memory. and pointers (tChar Font_array) to them to do Null or to the first character.

riuson, 2016-03-17:
@0xb0ba, не могли бы повторить вопрос на русском?

0xb0ba, 2016-03-17:
я ведь уверен был, что автор знает русский ))))

riuson, 2016-03-17:
А вы нашли программу не через русские форумы?)

Такая возможность была с самого начала. Просто не указывайте символы при создании шрифта, и они не будут включены в результат.

0xb0ba, 2016-03-17:
где-то в гугле нашел. сходу уже не вспомню.

это я понимаю, что можно вообще не добавлять ненужные символы. но тогда в Font_array будет лежать колво указателей равное добавленным символам, "дырок"  там не будет.

riuson, 2016-03-17:
В font_array будет указатели только на те символы, что были добавлены.
Вам чтоли нужны именно нулевые указатели на все незадействованные символы?

0xb0ba, 2016-03-17:
в промежутке между задействованными

riuson, 2016-03-17:
Смысл? Выборка из таблицы по индексу?

0xb0ba, 2016-03-17:
да, по индексу.
смысл в стандартной кодировке, но экономии памяти.

riuson, 2016-03-17:
Это слегка не вписывается. Есть же бинарный поиск. Тем более, количество кодов небольшое.
Могу предложить объединение указателей на идентичные символы.
Или нулевые указатели на символы, которые по размеру меньше указанных.

0xb0ba, 2016-03-17:
>  Есть же бинарный поиск

медленно, опять же память.

>Могу предложить объединение указателей на идентичные символы

не совсем понял, но в идеале хотелось бы совмещать одинаковые символы по начертанию из разных кодировок.

> Или нулевые указатели на символы, которые по размеру меньше указанных.

я пробовал делать символы с нулевыми размерами :)

кстати, символ с размерами 1x1 вдруг начинает занимать 2 байта.

riuson, 2016-03-17:
> в идеале хотелось бы совмещать одинаковые символы по начертанию из разных кодировок.

Да, именно это.

> я пробовал делать символы с нулевыми размерами :)

Сейчас это тоже не реализовано.

> кстати, символ с размерами 1x1 вдруг начинает занимать 2 байта.

Работа  изображениями меньше двух точек не предусмотрена.

0xb0ba, 2016-03-17:
> Сейчас это тоже не реализовано.

имхо проще всего :)

>Работа изображениями меньше двух точек не предусмотрена.

2x1 тоже два байта, впрочем это не важно.

riuson, 2016-03-17:
Как реализую, сообщу здесь.

0xb0ba, 2016-03-17:
спасибо!

riuson, 2016-03-19:
Итак...
- [x] 05135a179eecec8c4b9801566499d00180c8f8ac Совмещение символов одинакового содержимого/размера.

Символы не удаляются самой программой. Но в теги добавлены (см. 7feee224a6ce2602b4b03d2405bf1c5dba1c8ccb) 2 новых:
  * **out_char_code_sim** - код совпадающего символа или пустая строка;
  * **out_char_text_sim** - текст совпадающего символа или пустая строка.

Если добавить соответствующие строки шаблона для препроцессора (см. da07d4d938ddedd12a6a830a7fd492f2b2fbd22d), то после компиляции дублирующие данные будут опущены, а указатель будет соответствовать первой копии данных символа.

- [x] ef26c71fd636afb7c219c77f6ad5b0272636f77b Очистка данных символа (нулевые размеры и пустой массив), если символ не был преобразован.

NULL вставлять не буду в самой программе. Для этого можно применить тот же препроцессор.

- [x] Символ с размерами 1x1 занимает 0 байт.

Работа с символами меньшими **или равными** 2 точек не предусмотрена.
Также, изменив пользовательский алгоритм на что-то такое
~~~
if (image.width > 3) {
    for (var y = 0; y < image.height; y++) {
        for (var x = 0; x < image.width; x++) {
            image.addPoint(x, y);
        }
    }
}
~~~
можно пропустить создание массивов для отфильтрованных таким образом символов.
При передаче в шаблон у них будут, как указано выше, нулевые размеры и пустой массив.

Скачать новую версию можете здесь: [Download](http://www.riuson.com/lcd-image-converter/download)

0xb0ba, 2016-03-19:
оперативно!

>препроцессора 

исходник перестает быть читаем :(

`static const tChar Font_array[] PROGMEM = {
#if (0x0 == 0x0)
#if (0x04 != 0x0)
  {image_data_Font_0x30, 4},      // character: '0' 
#else
  {0, 4},      // character: '0' 
#endif
#else
#if (0x04 != 0x0)
  {image_data_Font_0x, 4},      // character (simulation): '' 
#else
  {0, 4},      // character (simulation): '' 
#endif
#endif

#if (0x0 == 0x0)
#if (0x04 != 0x0)
  {image_data_Font_0x31, 4},      // character: '1' 
#else
  {0, 4},      // character: '1' 
#endif
#else
#if (0x04 != 0x0)
  {image_data_Font_0x, 4},      // character (simulation): '' 
#else
  {0, 4},      // character (simulation): '' 
#endif
#endif
`

в итоге выходит NULL не нужен. все пропущенные символы можно сделать одним узнаваемым очертанием и получить на экране понятную картину.

riuson, 2016-03-19:
> исходник перестает быть читаем :(

Исходник для компилятора, не забывайте. А программно генерируемый исходник - особенно.
И у Вас что-то слишком сложно получилось. Посмотрите пример по ссылке ещё.

p.s. это не симуляция (simulation)...

0xb0ba, 2016-03-19:
>Исходник для компилятора, не забывайте.

ну от этого он не становится более читаем :)

>И у Вас что-то слишком сложно получилось. 

можно упростить, если учесть, что пустой и пустой похожий суть одно и тоже. но не сильно :)

>p.s. это не симуляция (simulation)...

один символ симулируется другим.

closed 2016-03-24 by riuson