riuson/lcd-image-converter #41: multi row per byte

mike603, 2016-04-01:
I am using the UltraChip UC1618t in my application for a 256 x 128 2 bit per pixel gray scale display.  Each bytes sequentially in memory represents a single pixel on four rows as follows (this is a bad 4x4 letter A in black and white).

As normal bit map (4x4):     0x0C, 0xC3, 0xFF, 0xC3
What the controller wants:  0xFC, 0x30, 0x30, 0xFC

As pixels
      X
  X    X
  XXXX
  X    X

In other words byte 1 bits 0 & 1 are row 0, column 0, bits 2 & 3 are row 1. column 0, bits 4 & 5 are row 2, column 0 and bits 6 & 7 are row 3, column 0.  Byte 2 bits 0 & 1 are row 0, column 1, bits 2 & 3 are row 1. column 1, bits 4 & 5 are row 2, column 1 and bits 6 & 7 are row 3, column 1.

How do I configure the image converter matrix for this?

Thank you,

           Mike

riuson, 2016-04-01:
Hello,
Try this settings:
[image]

[image]

[image]

mike603, 2016-04-01:
> Hello,
> Try this settings:
> 2016-04-02_01-41-02
> <https://cloud.githubusercontent.com/assets/415202/14219668/02304e16-f875-11e5-9574-4ca0f12a149c.png>
>
> 2016-04-02_01-43-56
> <https://cloud.githubusercontent.com/assets/415202/14219670/09a43112-f875-11e5-895a-625c1259de17.png>
>
> 2016-04-02_01-44-42
> <https://cloud.githubusercontent.com/assets/415202/14219673/0c46323a-f875-11e5-9349-781af05e1012.png>

Thank you, I will try that on Monday.

mike603, 2016-04-04:
Vladimir,

Thank you for your help.  It didn't quite work.

I have attached a sample image. Below is the array that I hand converted.

In this particular case I am just converting text so I am converting 
from a 1bit PNG file.

It is possible that I might be converting from 4 color grayscale in an 
8-bit png or gif.

     //
     //  Rows 0 - 3
     //
     0x00, 0x00, 0xc0, 0xfc, 0x3c, 0xfc, 0xc0, 0x00, 0x00, 0xfc, 0xfc, 
0x0c, 0x0c, 0xfc, 0xf0, 0x00,
     0xc0, 0xf0, 0x3c, 0x0c, 0x0c, 0x3c, 0x30, 0x00, 0xfc, 0xfc, 0x0c, 
0x0c, 0xfc, 0xf0, 0x00,
     //
     //  Rows 4 - 7
     //
     0x00, 0xf0, 0xff, 0x3f, 0x30, 0x3f, 0xff, 0xf0, 0x00, 0xff, 0xff, 
0x03, 0x03, 0xff, 0xfc, 0x00,
     0x3f, 0xff, 0xc0, 0x00, 0x00, 0xc0, 0xc0, 0x00, 0xff, 0xff, 0x00, 
0x00, 0xff, 0xff, 0x00,
     //
     //  Rows 8 - 11
     //
     0x00, 0x03, 0x03, 0x00, 0x00, 0x00, 0x03, 0x03, 0x00, 0x03, 0x03, 
0x03, 0x03, 0x03, 0x00, 0x00,
     0x00, 0x00, 0x03, 0x03, 0x03, 0x03, 0x00, 0x00, 0x03, 0x03, 0x03, 
0x03, 0x03, 0x00, 0x00

Thank you so much for your help,

               Mike K.

riuson, 2016-04-04:
Mike, you are writing to the issue tracking system on Github via email.
All attached files are lost.
Please login on GitHub and attach them here, or email to me directly.

mike603, 2016-04-04:
Vladimir,

Thank you for your time.  Here is the missing attachment.
[image]

mike603, 2016-04-04:
Sorry, I didn't mean to close this.  Forgive my GIT ignorance.

riuson, 2016-04-04:
Ok. Are was invalid bits order in first message?

Are this valid ? -
byte 1
bits 0 & 1 are row **3**, column 0;
bits 2 & 3 are row **2**. column 0;
bits 4 & 5 are row **1**, column 0;
bits 6 & 7 are row **0**, column 0.

Or you just have reordered data bits between mcu and display?

[image]

mike603, 2016-04-04:
> 2016-04-04_22-02-18

Vladimir,

Bits 6 and 7 are Row 3, column 0
Bits 4 and 5 are Row 2, column 0
Bits 2 and 3 are Row 1, column 0
bits 0 and 1 are Row 0, column 0

I will try your new idea later today.

Thank you again,

              Mike K.

riuson, 2016-04-04:
I use following assumptions:
Bit 0 - least significant bit, bit 7 - most significant bit. 76543210.
Point (**0, 0**) - left top corner. Point (last, last) - right bottom corner.
Row 0 - top (up) row. Last row - bottom (down) row.

mike603, 2016-04-04:
Vladimir,

That worked perfectly, thank you.

One suggestion for a future version of this program.  In situations like 
this one, the "Split to rows" on the Image tab generates one entry per 
line or all entries on a single line.  It might be nice to be able to 
format the output by sets of rows (int this case 4).

Thank you again, this was very helpful.

            Mike

Below is the xml file for the UltraChip UC1618T LCD Controller:

<?xml version="1.0" encoding="utf-8"?>
<preset version="3">
   <name>Grayscale 2</name>
   <prepare>
     <convType>1</convType>
     <monoType>1</monoType>
     <edge>128</edge>
     <scanMain>2</scanMain>
     <scanSub>0</scanSub>
     <inverse>1</inverse>
     <bandScanning>1</bandScanning>
     <bandWidth>4</bandWidth>
     <useCustomScript>1</useCustomScript>
     <customScript><![CDATA[/*
  * left to right
  * forward
  * bands
  */

var bandY = 0;

do {
     for (var x = 0; x < image.width; x++) {
         for (var y =  image.bandSize - 1; y>=0 ; y--) {
             image.addPoint(x, bandY + y);
         }
     }
     bandY += image.bandSize;
} while (bandY < image.height);]]></customScript>
   </prepare>
   <matrix>
     <maskUsed>00000003</maskUsed>
     <maskAnd>ffffffff</maskAnd>
     <maskOr>00000000</maskOr>
     <maskFill>0000ffff</maskFill>
     <operations count="1">
       <operation index="0">
         <mask>000000c0</mask>
         <shift>6</shift>
         <left>0</left>
       </operation>
     </operations>
   </matrix>
   <reordering>
     <operations count="0"/>
   </reordering>
   <image>
     <bytesOrder>0</bytesOrder>
     <blockSize>0</blockSize>
     <blockDefaultOnes>0</blockDefaultOnes>
     <splitToRows>0</splitToRows>
     <compressionRle>0</compressionRle>
     <compressionRleMinLength>2</compressionRleMinLength>
     <blockPrefix>0x</blockPrefix>
     <blockSuffix></blockSuffix>
     <blockDelimiter>, </blockDelimiter>
   </image>
   <font>
     <bom>0</bom>
     <sortOrder>1</sortOrder>
     <codec>ASMO-708</codec>
   </font>
   <templates>
     <images>:/templates/image_convert</images>
     <fonts>:/templates/font_convert</fonts>
   </templates>
</preset>

riuson, 2016-04-04:
> One suggestion for a future version of this program. In situations like this one, the "Split to rows" on the Image tab generates one entry per line or all entries on a single line. It might be nice to be able to format the output by sets of rows (int this case 4).

Feature "Split to rows" means completely different than placing one item per line.
But I will think on the limitation of the line length.

Xml with syntax highlight:
~~~xml
<?xml version="1.0" encoding="utf-8"?>
<preset version="3">
   <name>Grayscale 2</name>
   <prepare>
     <convType>1</convType>
     <monoType>1</monoType>
     <edge>128</edge>
     <scanMain>2</scanMain>
     <scanSub>0</scanSub>
     <inverse>1</inverse>
     <bandScanning>1</bandScanning>
     <bandWidth>4</bandWidth>
     <useCustomScript>1</useCustomScript>
     <customScript><![CDATA[/*
  * left to right
  * forward
  * bands
  */

var bandY = 0;

do {
     for (var x = 0; x < image.width; x++) {
         for (var y =  image.bandSize - 1; y>=0 ; y--) {
             image.addPoint(x, bandY + y);
         }
     }
     bandY += image.bandSize;
} while (bandY < image.height);]]></customScript>
   </prepare>
   <matrix>
     <maskUsed>00000003</maskUsed>
     <maskAnd>ffffffff</maskAnd>
     <maskOr>00000000</maskOr>
     <maskFill>0000ffff</maskFill>
     <operations count="1">
       <operation index="0">
         <mask>000000c0</mask>
         <shift>6</shift>
         <left>0</left>
       </operation>
     </operations>
   </matrix>
   <reordering>
     <operations count="0"/>
   </reordering>
   <image>
     <bytesOrder>0</bytesOrder>
     <blockSize>0</blockSize>
     <blockDefaultOnes>0</blockDefaultOnes>
     <splitToRows>0</splitToRows>
     <compressionRle>0</compressionRle>
     <compressionRleMinLength>2</compressionRleMinLength>
     <blockPrefix>0x</blockPrefix>
     <blockSuffix></blockSuffix>
     <blockDelimiter>, </blockDelimiter>
   </image>
   <font>
     <bom>0</bom>
     <sortOrder>1</sortOrder>
     <codec>ASMO-708</codec>
   </font>
   <templates>
     <images>:/templates/image_convert</images>
     <fonts>:/templates/font_convert</fonts>
   </templates>
</preset>
~~~

mike603, 2016-04-04:
Vladimir,

I agree, and I understand that split to rows does not mean one per line, 
however, in this situation that is what is causes.

Thanks again,

              Mike

closed 2016-04-09 by riuson