riuson/lcd-image-converter #42: descent parameter

sumotoy, 2016-06-04:
It's a request, not a bug!

It's possible access the 'descent' parameter from baseline in template?
Like this:
[image]

riuson, 2016-06-04:
So as it available in [font metrics](http://doc.qt.io/qt-5.6/qfontmetrics.html), then yes.
But font's modification can change its to wrong value.

riuson, 2016-06-07:
Tags `fnt_descent` and `fnt_ascent` added in rev. 07451b763aa7769554a547d348ca0aad46caee6e from 2016-06-07.

sumotoy, 2016-06-07:
> font's modification can change its to wrong value.
You right. I've compiled https://github.com/riuson/lcd-image-converter/commit/07451b763aa7769554a547d348ca0aad46caee6e and tested. It works!
I've noticed that optimixe height have strong impact on ascent/descent. 
Cutting pixel from top should subtract from ascent and from bottot to descent

riuson, 2016-06-08:
Because after getting images for characters we have just images. Every image can be modified **independently**, by width and height. So common 'descent' property of source font become invalid.

sumotoy, 2016-06-08:
I understand, even if I never touched a single height but I always use Optimize height...
Can be useful detect single height change and drop the ascent/descent value to 0, as opposite (if no single height change detected) track the Optimize height... changes and update ascent/descent values?
Like this it's possible use ascent/descent when one of the 2 values is not 0.

riuson, 2016-06-09:
I think it's possible. But you have to wait much longer time than that taken by implementation of previous feature.

sumotoy, 2016-06-09:
No rush riuson, this app is still the best one I ever find! I'm actually including link in all display libraries!

riuson, 2016-06-22:
Try compile branch `feature-operations`, revision 153bc449d399ea94ae1afa303e1b32e794d0f012 .

sumotoy, 2016-06-23:
Amazing! Thanks, I will try this afternoon, can't wait!

riuson, 2016-06-27:
I have uploaded compiled binary rev. aeaec30c605c26a2120fc3fd7643d96b8050e187 from 2016-06-27.

sumotoy, 2016-06-27:
* ascent:		12
* descent:		3

You got it! It works! Obviously it doesn't work for old stored fonts but I have converted new one and it works!
I have tried optimized height and correctly updated descent-ascent!
Did you save the original font ascent/descent data in the saved font file? (NOT the converted one, I know it's present)

riuson, 2016-06-28:
I save modified data in xml font file:
```
    <size>-1</size>
    <ascent>12</ascent>
    <descent>3</descent>
    <style>Normal</style>
```
This values can be changed with font resize/minimize. So it is not original values of the font.

closed 2016-07-22 by riuson