riuson/lcd-image-converter #45: Cannot build on Mac OS 10.12.4

sbfreddie, 2017-04-16:
Hello:
While trying to build this app using qmake according to your instructions, qmake gives this error:
RCC: Error in 'resources/resources.qrc': Cannot find file 'lcd-image-converter-ru.qm'

What am I doing wrong?

Thanks,
Ed

riuson, 2017-04-17:
Hello.
Its OK. This file compiles from lcd-image-converter-ru.qm with `make` command.
For example, see at lines 397 and 403 from [this build log](https://travis-ci.org/riuson/lcd-image-converter/jobs/213045850).
This file (qm) not included because it is binary file what often changed.

closed 2017-05-02 by riuson

dimka-rs, 2019-03-29:
Same problem with linux:
`RCC: Error in 'resources/resources.qrc': Cannot find file 'lcd-image-converter-ru.qm'`
How do you build app correctly then?

riuson, 2019-03-29:
This message appears after the start of the qmake?
Have you tried running make after this?

dimka-rs, 2019-03-29:
Agh, got it, Makefile is created anyway. Probably you should mention in readme that this error is safe to ignore. Thank you!

riuson, 2019-03-29:
This was mentioned in this topic above and docs: https://www.riuson.com/lcd-image-converter/others/build-from-sources