riuson/lcd-image-converter #46: Forced font reordering

zoon81, 2017-11-30:
Hi,
I made a new font table with only number and a plus sign symbol and negative sign symbol (0,1,2,3,4,5,6,7,8,9,+,-) in this order.
But when I am hit convert and I check the output, the order of the characters are changed (+,-,0,1,2,3,4,5,6,7,8,9). This is behave like an ASCII value based order.
I builded from source under mac in yesterday.
Revision bf3294d from 2017-03-20 21:03:09 +0500
I linked the generated .c file. It is list in the right order in the description of the file but wrong order in the rest.
[https://pastebin.com/J6YqsfGe](generated_file)

riuson, 2017-12-21:
You can disable character's sorting:
[image]

zoon81, 2017-12-23:
I missed this one.
Many thanks!

closed 2017-12-23 by zoon81