riuson/lcd-image-converter #47: Output file encoding

antoniovazquezblanco, 2017-12-16:
If in my templates I use special characters, output files get wrongly encoded characters. For example, if I include the character `á` then in the output source code I get `Ã¡`.

riuson, 2017-12-16:
Please attach files:
- font's xml document
- font's template
- preset's export
- output file

antoniovazquezblanco, 2017-12-16:
Template:
```
$(start_block_header)
/**
 * @file    $(doc_name_ws).c
 * @author  Antonio Vázquez Blanco
 *
 * Automatically generated file from image using LCD Image Converter.
 *
 */

 
#include "../ch-devices-drivers/gdew0154t8.h"
$(end_block_header)

$(start_block_images_table)
static const uint$(img_data_block_size)_t image_data_$(doc_name_ws)[$(out_blocks_count)] = {
    $(out_image_preview)
    $(out_image_data)
};

const image_t image_$(doc_name_ws) = {
	image_data_$(doc_name_ws),
	$(out_image_width),
	$(out_image_height),
    $(img_data_block_size)
};
$(end_block_images_table)
```
Default font template.
Presets:
```
<?xml version="1.0" encoding="utf-8"?>
<preset>
  <name>CarrierPresets</name>
  <prepare>
    <convType>0</convType>
    <monoType>1</monoType>
    <edge>128</edge>
    <scanMain>0</scanMain>
    <scanSub>0</scanSub>
    <inverse>1</inverse>
    <bandScanning>0</bandScanning>
    <bandWidth>1</bandWidth>
    <useCustomScript>0</useCustomScript>
    <customScript><![CDATA[]]></customScript>
    <customPreprocessScript><![CDATA[]]></customPreprocessScript>
  </prepare>
  <matrix>
    <maskUsed>00000001</maskUsed>
    <maskAnd>ffffffff</maskAnd>
    <maskOr>00000000</maskOr>
    <maskFill>000000ff</maskFill>
    <operations count="2">
      <operation index="0">
        <mask>ff000000</mask>
        <shift>0</shift>
        <left>0</left>
      </operation>
      <operation index="1">
        <mask>00000001</mask>
        <shift>0</shift>
        <left>0</left>
      </operation>
    </operations>
  </matrix>
  <reordering>
    <operations count="0"/>
  </reordering>
  <image>
    <bytesOrder>0</bytesOrder>
    <blockSize>0</blockSize>
    <blockDefaultOnes>0</blockDefaultOnes>
    <splitToRows>1</splitToRows>
    <compressionRle>0</compressionRle>
    <compressionRleMinLength>2</compressionRleMinLength>
    <blockPrefix><![CDATA[0x]]></blockPrefix>
    <blockSuffix><![CDATA[empty]]></blockSuffix>
    <blockDelimiter><![CDATA[, ]]></blockDelimiter>
    <previewPrefix><![CDATA[// ]]></previewPrefix>
    <previewSuffix><![CDATA[empty]]></previewSuffix>
    <previewDelimiter><![CDATA[empty]]></previewDelimiter>
    <previewLevels><![CDATA[ 
•]]></previewLevels>
  </image>
  <font>
    <bom>0</bom>
    <sortOrder>1</sortOrder>
    <codec>ASMO-708</codec>
  </font>
  <templates>
    <images>D:/Projects/ch-devices-sw-carrier/carrier.X/images/image.tmpl</images>
    <fonts>:/templates/font_convert</fonts>
  </templates>
</preset>
```
Output file header:
```

/**
 * <value not defined>.c
 * @author  Antonio VÃ¡zquez Blanco
 *
 * Automatically generated file from image using LCD Image Converter.
 *
 */

 ```

antoniovazquezblanco, 2017-12-16:
Awesome response speed! Thank you! ;)

riuson, 2017-12-16:
Try save template file as UTF-8 **with BOM**.

antoniovazquezblanco, 2017-12-16:
That worked! Thank you very much!

closed 2017-12-16 by antoniovazquezblanco