riuson/lcd-image-converter #52: Can't built on linux, installed lots of qt5 dependencies..

salmansheikh, 2020-02-16:
qmake error:

lcd-image-converter-master$ qmake
RCC: Error in 'resources/resources.qrc': Cannot find file 'lcd-image-converter-ru.qm'

graemechapman, 2020-02-17:
Same issue, removing reference to the translation files worked for me (although obviously you then lose support for Russian)
https://github.com/riuson/lcd-image-converter/blob/master/lcd-image-converter.pro#L418-L423
https://github.com/riuson/lcd-image-converter/blob/master/resources/resources.qrc#L13-L15

riuson, 2020-04-05:
This is the **intended** behavior.
https://www.riuson.com/lcd-image-converter/others/build-from-sources

dsalychev, 2020-07-22:
btw, I've managed to build the converter on FreeBSD with Xfce. I could prepare an instruction how to do so. Is anyone interested?