riuson/lcd-image-converter #56: Patch with "out_blocks_offset" tag

dsalychev, 2020-07-12:
Hi,

You'll find a patch which adds a new "out_blocks_offset" tag to the templates in the archive. It might be useful to provide an offset in a single image data array (see TestFont.c and TestFont2.c with their _glyphs and _bitmaps variables) to a glyph:
```
/* Glyphs bitmap */
static const uint8_t PROGMEM _bitmaps[] = {
	0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
	0x00, 0x00, 0xe0, 0x18, 0x00, 0x00, 0x0c, 0x01, 0x00, 0x00,
	0x00, 0x38, 0x00, 0x38, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        ...
};

/* Glyphs */
static const MSIM_SH1106Glyph_t _glyphs[] = {
	{ .code=0x20, .data=&_bitmaps[0], .width=4, .height=15, .data_size=8, },	/* ' ' */
	{ .code=0x21, .data=&_bitmaps[8], .width=5, .height=15, .data_size=8, },	/* '!' */
	{ .code=0x22, .data=&_bitmaps[18], .width=6, .height=15, .data_size=8, },	/* '\"' */
        ...

/* Font */
const MSIM_SH1106Font_t XG_FONT_TestFont = {
	.length = 95,
	.glyphs = _glyphs,
};
```
This is how I generate fonts for my tamagotchi-like toy.

[out_blocks_offset_patch.zip](https://github.com/riuson/lcd-image-converter/files/4908907/out_blocks_offset_patch.zip)

dsalychev, 2020-07-15:
@riuson Any thoughts?

riuson, 2020-07-15:
@dsalychev , it is useful, accepted. Thanks.

riuson, 2020-07-16:
Added in rev. d56faf9

dsalychev, 2020-07-16:
Looks good to me :) I tried to build and run it with my font template - no problem. The only adjustment is for the tag's name itself needed. Thanks for accepting.

closed 2020-07-16 by dsalychev