riuson/lcd-image-converter #69: Adafruit GFX Compatibility

boromyr, 2021-06-14:
Hi, I wanted to know if you can export fonts for compatibility with the Adafruit GFX library, I tried to look for a template but I couldn't find it. This is an example of[font Adafruit](https://github.com/Bodmer/TFT_eSPI/blob/master/Fonts/GFXFF/FreeMono12pt7b.h). Everything is exported as a vertical image, and after those that should be the coordinates, but I can't reconstruct the template.

riuson, 2021-06-20:
Hello,
Try the following template.
It is highly likely that it will not give you an immediately working result. But it can give you a starting point.
```
$(start_block_header)
#include <stdint.h>

//#define XADVANCE ($(out_images_max_width)+1)
#define YADVANCE ($(out_images_max_height)+1)

$(end_block_header)

const uint8_t $(doc_name_ws)_Bitmaps[] PROGMEM = {
$(start_block_images_table)
  $(out_image_data)
  #if ($(out_blocks_count) > 0)
  $(out_comma)
  #endif
$(end_block_images_table)
};

const GFXglyph $(doc_name_ws)_Glyphs[] PROGMEM = {
$(start_block_images_table)
  //Index,  W, H,xAdv,dX, dY
  { $(out_char_offset), $(out_char_image_width), $(out_char_image_height), /*XADVANCE*/ $(out_image_width) + 1, $(out_char_image_left), ( $(out_image_height) - ($(out_char_image_top) + $(out_char_image_height)) )}$(out_comma) // 
$(end_block_images_table)
};

$(start_block_images_table)
#ifndef FIRST_CHAR_CODE
#define FIRST_CHAR_CODE 0x$(out_char_code)
#endif

#ifdef LAST_CHAR_CODE
#undef LAST_CHAR_CODE
#endif

#define LAST_CHAR_CODE 0x$(out_char_code)

$(end_block_images_table)

const GFXfont $(doc_name_ws) PROGMEM = {
  (uint8_t  *)$(doc_name_ws)_Bitmaps,
  (GFXglyph *)$(doc_name_ws)_Glyphs,
  FIRST_CHAR_CODE, LAST_CHAR_CODE, YADVANCE
};

#undef XADVANCE
#undef YADVANCE
#undef FIRST_CHAR_CODE
#undef LAST_CHAR_CODE
```

boromyr, 2021-06-22:
Wow, thank you very much, I'll try it and I'll try to fix it. You could add it to the repository, so they can use it too, almost all library use Adafruit GFX and there are no simple and effective tools for fonts.
Excellent app!!!!

riuson, 2021-06-23:
If you manage to bring the template to working condition, let me know. Maybe I will add it to the package.

closed 2021-09-29 by riuson