riuson/lcd-image-converter #71: need help

discip, 2021-09-28:
Good evening Sir.
First of all, thank you for providing this very helpful software! 👍

I am having some trouble understanding what to change to convert this:
[image]

to this:
```cpp

0x00, 0xE0, 0x18, 0x04, 0x02, 0x02, 0x01, 0x41, 0x61, 0x61, 0x61, 0xE1, 0xC1, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81,
0xC1, 0xE1, 0x61, 0x61, 0x61, 0x41, 0x01, 0x01, 0x02, 0x02, 0x04, 0x18, 0xE0, 0x00, 0x00, 0xE0, 0x18, 0x04, 0x02, 0x02, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01,
0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x99, 0x65, 0x01, 0x01, 0x81, 0x41, 0x01, 0x02, 0x02, 0x04, 0x18, 0xE0,
0x00, 0x07, 0x18, 0x20, 0x40, 0x40, 0x80, 0x82, 0x86, 0x86, 0x86, 0x87, 0x83, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81, 0x81,
0x83, 0x87, 0x86, 0x86, 0x86, 0x82, 0x80, 0x80, 0x40, 0x40, 0x20, 0x18, 0x07, 0x00, 0x00, 0x07, 0x18, 0x20, 0x40, 0x40, 0x80, 0x82, 0x87, 0x85, 0x85, 0x85, 0x85, 0x87,
0x87, 0x85, 0x87, 0x85, 0x87, 0x87, 0x82, 0x82, 0x82, 0x80, 0x82, 0x80, 0x82, 0x82, 0x82, 0x92, 0x8A, 0x84, 0x82, 0x81, 0x80, 0x80, 0x80, 0x40, 0x40, 0x20, 0x18, 0x07};
```

thanks in advance
kind regards

riuson, 2021-09-29:
Hello,
Unfortunately, I'm not a cryptanalyst.
Conversion requires knowing the format of the output file.

discip, 2021-09-29:
@riuson
Good afternoon sir,
first of all thank you for having reacted to this post. 😃👍
Please don't be offended by my approach, since I am not familiar with this kind of topic, I thought the ***'source'*** and the ***'result'*** would suffice. 😁
I used your tool to convert the picture above but was not able to get it done no matter what I choose, it was almost the same but not exactly.
What exactly do you mean by *format*? 😅

The things I know are:

1. monochrome
2. 8 bit
3. 84 x 16
4. left to right

thanks in advance

riuson, 2021-09-29:
*Data format*.
Description of each byte in resulting array.

Are you have a code what draws image from this array?

discip, 2021-09-29:
It is used in this software: https://github.com/Ralim/IronOS/blob/master/source/Core/Drivers/Font.h
But unfortunately I do not know, which part is responsible for decoding the arrays. 😅

riuson, 2021-09-29:
Can you provide rendered images for `☐`, `.`, `|` ?

discip, 2021-09-29:
I have compared the two arrays mentioned above and realized: The code itself is the same except for the fact that it is arranged somewhat differently:
To put it the way I would need it I would have to read the code (which was generated by your tool)

from bottom right to the top right and than
from the bottom left to the top left

and afterwards write it down

from left to right and top to bottom
divided into 6 equal chunks. Hopefully I made myself clear enough. 😅

discip, 2021-09-29:
Sorry, but I can't since I am only seeing the rendered images on the Iron.

riuson, 2021-09-29:
Ok, i understand format...
`☑`

```
0011111111111100
0100000000000010
0100001000000010
0100010000000010
0100110000000010
0101100000000010
0100011000000010
0100000110000010
0100000001100010
0100000000011010
0100000000000010
0011111111111100
```

riuson, 2021-09-29:
The image you provided in first message does not match the byte array. It is only slightly similar.

I have attached preset what you can import in app's options dialog - [discip.zip](https://github.com/riuson/lcd-image-converter/files/7253595/discip.zip)

Demo:
[image])
https://github.com/Ralim/IronOS/blob/07ceb46d8479f802cd29d8fa67993e4b379b669d/source/Core/Drivers/Font.h#L40
```
0xFC, 0x02, 0x02, 0x02, 0x02, 0x02, 0x02, 0x82, 0x62, 0x1A, 0x02, 0xFC, 0x3F, 0x40, 0x42, 0x46, 0x4C, 0x58, 0x46, 0x41, 0x40, 0x40, 0x40, 0x3F, // ☑ (check box on, menu true)
```

discip, 2021-09-29:
@riuson
Maybe this could be of any help:
https://github.com/Ralim/IronOS/blob/master/Translations/BitmapEditor.html

riuson, 2021-09-30:
Script inside this html page is not working in my browser.

[image]

discip, 2021-09-30:
I think I am close to the solution with this:
[IronOS _almost.txt](https://github.com/riuson/lcd-image-converter/files/7261289/IronOS._almost.txt)

The *only* additional steps I have to perform are:
1. ***reverse line order***
2. ***delete carriage return and line feed***
3. ***insert the carriage return and line feed*** after every 28th **`,`** (comma)

Can this be achieved with your software as well?

riuson, 2021-09-30:
1. What is *reverse line order*?
2. May be, depends on the cause of the occurrence. But compiler is indiffirent to this.
3. No. But compiler is indiffirent to this.

Above, I gave the settings that produce the same result as specified in the project. Why aren't you using it?

discip, 2021-09-30:
1. Ah sorry I used the term notepad++ uses to swap the first line with the last, the second with the forelast ...

> Above, I gave the settings that produce the same result as specified in the project. Why aren't you using it?

But you said yourself that this setting is not applicable to the image I posted initially.

> The image you provided in first message does not match the byte array. It is only slightly similar.

riuson, 2021-09-30:
> Ah sorry I used the term notepad++ uses to swap the first line with the last, the second with the forelast ...

This swapping is already enabled in my preset.

***

> But you said yourself that this setting is not applicable to the image I posted initially.

Yes. Because your image is seems not correctly encoded for algorithm from here:

> It is used in this software: https://github.com/Ralim/IronOS/blob/master/source/Core/Drivers/Font.h

Difference seems like missing lines on border. You can decode your byte array by may pictire posted above.

discip, 2021-09-30:
@riuson
# Again a big thank you for this software and the effort you took on to help me figuring this out! ❤👍

Here is the setting attached that did the trick *(The only thing I have to do manually, is to replace the ***white space*** with ***carriage return & line feed*** after every 28th **`,`**)* 😅
[84x16_IronOS.txt](https://github.com/riuson/lcd-image-converter/files/7261906/84x16_IronOS.txt)

closed 2021-09-30 by discip

riuson, 2021-10-01:
> 84x16_IronOS.txt

There is scanning from latest column to first. Image is flipped horizontally.

> replace the white space with carriage return & line feed after every 28th

This is completely unnecessary. The compiler doesn't care about the length of the string.
If you are worried about the appearance of the code, you can use utilities like [AStyle](http://astyle.sourceforge.net/). All code of the lcd-image-converter is formatted using that utility.