riuson/lcd-image-converter #75: Wrong version gets downloaded as latest on sourceforge.com

kenyerman, 2022-05-15:
## Steps to reproduce

1. Open the project's _sourceforge_ page: [https://sourceforge.net/projects/lcd-image-converter/](https://sourceforge.net/projects/lcd-image-converter/)
2. Click the big green ___Download___ CTA button

## What happens
`lcd-image-converter-20161012.zip` gets downloaded.

## What should happen
`lcd-image-converter-20190317.zip` should get downloaded.

## Additional info
When I open the [Files page on sourceforge](https://sourceforge.net/projects/lcd-image-converter/files) I can download `lcd-image-converter-20190317.zip` but there is also a CTA button saying that the latest version is `lcd-image-converter-20161012.zip`.

## Why is this a problem
Your users don't get the latest updates of the software.

### Personal example:
I've almost opened an issue on a non-functional console parameter, but when I've reached the version part of the issue, I realized that an old version was in use.

riuson, 2022-05-17:
Thanks. Fixed.

kenyerman, 2022-05-18:
### Verified!
Now when I click the button, the `lcd-image-converter-20190317.zip` is downloaded.

Also I want to thank you for this software. It is very useful and works well. 💖🎉

__Closing because issue solved.__

closed 2022-05-18 by kenyerman