riuson/lcd-image-converter #79: Fails to build as

DavidJRichards, 2023-03-15:
Build fails with message: 'Project ERROR: Unknown module(s) in QT: xmlpatterns'

Link here shows alternative build method which worked for me: https://askubuntu.com/questions/1418672/qmake-problem-with-image-converter-on-22-04

```
git clone https://github.com/riuson/lcd-image-converter.git
cd lcd-image-converter/
git checkout develop

sudo apt install qtcreator qtbase5-dev qt5-qmake cmake
sudo apt install build-essential mesa-common-dev
sudo apt install libqt5svg5-dev libqt5xmlpatterns5-dev qtdeclarative5-dev

qmake
make -j 20
./release/linux/output/lcd-image-converter
```

skeyvin, 2023-11-24:
`sudo apt install libqt5xmlpatterns5-dev`

riuson, 2024-06-20:
Info was added to https://lcd-image-converter.riuson.com/en/docs/others/build-from-sources/