riuson/lcd-image-converter #80: Byte order in image options not works

Depech101, 2023-03-18:
[image]
[image]
[image]
[image]
[image]
[image]
Output code doesent changes regardless of litleEndian i have choosen or BigEndian.

riuson, 2023-03-18:
Hello,
The 'Byte Order' option affects only multi-byte values (16, 32 bits).
8-bit values are encoded as is.

Depech101, 2023-04-30:
Sorry, i totally forgot why i needed to do this, may be for negative picture. I made it with a photo editor.