riuson/lcd-image-converter #81: lcd-image-converter

gordonpow, 2023-09-07:
Hello, I accidentally deleted the color r6g5b6 in the preset option of the operation. Where can I find this file?

riuson, 2023-09-07:
Hello, gordonpow.
 
Use  --config-initialize  command line argument:
Command line | LCD Image Converter (riuson.com) 

Вы писали 7 сентября 2023 г., 15:15:47:

> Hello, I accidentally deleted the color r6g5b6 in the preset option of the operation. Where can I find this file?

-- 
С уважением,
 Vladimir                         ***@***.***

riuson, 2023-09-07:
Hello,
Use  `--config-initialize`  [command line](https://lcd-image-converter.riuson.com/en/docs/others/command-line/) argument.

gordonpow, 2023-09-07:
Thank you, I have solved my problem, thank you for taking the time to reply
to my message.

closed 2023-09-27 by riuson