riuson/lcd-image-converter #83: Incorrect .ttf to monochrome 16-bit conversion

mithilshah3, 2024-06-14:
I'm trying to convert a .ttf file to an array of 16-bit words using 'monochrome' preset with character width set to 'proportional'. But the generated array size doesn't match the width of the character. For example, if the width & height of a character is 9 & 25 respectively, the generated array consists of 50 16-bit words, i.e., 2 words for each row when in fact it should contain only 25 16-bit words. The byte-order setting (endianness) has no effect on the array size. Is there some other setting that needs to be applied?

riuson, 2024-06-20:
First of all, the font size may not match the size of the resulting character in bit representation. After creating the font, it may be necessary to trim unused (empty) spaces around the characters.

In the second place, recent versions have added such an option. Try it:
[image]

Tombo89, 2025-09-15:
The used preset is monochrome

[Font.xml](https://github.com/user-attachments/files/22329872/Font.xml) and the xml file

closed 2025-09-15 by riuson