riuson/lcd-image-converter #85: A problem with $(out_image_width) and $(out_image_height) tags

teslasystemsx, 2024-07-15:
Hello.

I'm working on my own template for images, but faced the issue with "$(out_image_width)" and "$(out_image_height)" tags.

If I write a template like this

```
$(start_block_header)
#include <stdint.h>

#define Width $(out_image_width)
#define Height $(out_image_height)

$(end_block_header)

$(start_block_images_table)
static const uint$(img_data_block_size)_t image_data_$(doc_name_ws)[$(out_blocks_count)] = {
    $(out_image_preview)
    $(out_image_data)
};
const tImage $(doc_name_ws) = { image_data_$(doc_name_ws), $(out_image_width), $(out_image_height),
    $(img_data_block_size) };
$(end_block_images_table)
```

which contains the image width and height in defines section, there will be no actual values in output file. It will look like this:
```
#define Width <value not defined>
#define Height <value not defined>
```
But the constant at the very end of the file will contain these values as it should. It's strange. Can you fix this?

riuson, 2024-07-15:
Hello,
With fonts you can use:
- out_images_max_height
- out_images_max_width

Look at [templates](https://lcd-image-converter.riuson.com/en/docs/templates/) page.

teslasystemsx, 2024-07-15:
> Hello, With fonts you can use:
> 
>     * out_images_max_height
> 
>     * out_images_max_width
> 
> 
> Look at [templates](https://lcd-image-converter.riuson.com/en/docs/templates/) page.

Tried these tags and they work for images also, thank you!
But anyway it's strange, why the tags I mentioned above do work in one part of the template and at the same time they don't work in another part.

riuson, 2024-07-15:
out_image_width and out_image_height are valid for one image (image document).
A font document, on the other hand, contains many images (characters), each of which can have its own size. So these tags don't have values outside the context of a concrete character.

teslasystemsx, 2024-07-15:
> out_image_width and out_image_height are valid for one image (image document). A font document, on the other hand, contains many images (characters), each of which can have its own size. So these tags don't have values outside the context of a concrete character (inside loop `$(start_block_images_table)` ... `$(end_block_images_table)`).

Understood. Thanks for clarification.

closed 2024-07-15 by teslasystemsx