riuson/lcd-image-converter #87: Converting fonts to C yields a bunch of <value not defined> fields

OpusElectronics, 2024-12-03:
Context:
Revision 4d43872 from 2022-06-23 22:11:57 +0500
Qt 5.15.16
Linux

Monochrome fonts

All character images generate something like:
```
static const tImage Font_0x7d = { image_data_Font_0x7d,
    <value not defined>, <value not defined>, 8};
```

Likewise, font arrays have this kind of entries:
```
{0x21, &Font_0x21, <value not defined>, <value not defined>, 4, 20 },
```

What could be the cause?

OpusElectronics, 2024-12-03:
Here: 
[Noto-14.zip](https://github.com/user-attachments/files/17988701/Noto-14.zip)

riuson, 2024-12-03:
No, **template** file.
LIke
```
$(start_block_header)
#include <stdint.h>

//#define XADVANCE ($(out_images_max_width)+1)
#define YADVANCE ($(out_images_max_height)+1)

$(out_char_code_min)
$(out_char_code_max)
...............
```

riuson, 2024-12-03:
Yes, really...
In archive you can find **image.tmpl** and **font.tmpl** files.
Open Options - Conversion - Templates and select this files.
In **font.tmpl** replace **out_char_image_width** with **out_image_width** and **out_char_image_height** with **out_image_height** .

OpusElectronics, 2024-12-03:
Thanks! That does solve the issue with tImage, but I still have an issue with tChar, due to the **out_char_image_left** and **out_char_image_top**. I tried replacing them with **out_image_left** and **out_image_top** (thinking you may have renamed them similarly), but it doesn't work.

riuson, 2025-04-14:
[Available tags](https://lcd-image-converter.riuson.com/en/docs/templates/)

Template:
```cpp
  // out_char_image_left: $(out_char_image_left)
  // out_char_image_top: $(out_char_image_top)
  // out_char_image_width: $(out_char_image_width)
  // out_char_image_height: $(out_char_image_height)
  // out_image_width : $(out_image_width)
  // out_image_height : $(out_image_height)
```

Results:
```cpp
  // out_char_image_left: <value not defined>
  // out_char_image_top: <value not defined>
  // out_char_image_width: <value not defined>
  // out_char_image_height: <value not defined>
  // out_image_width : 21
  // out_image_height : 48
```

But with enabled *compact* feature:
[image]

```cpp
  // out_char_image_left: 1
  // out_char_image_top: 5
  // out_char_image_width: 18
  // out_char_image_height: 33
  // out_image_width : 21
  // out_image_height : 48
```